mdpnp/mdpnp #22: Place artifactory artifacts directly into the build tree

simonkelly1973, 2019-10-23:
We are migrating to a new server for the hosted elements of openice, such as build.openice.info .  It seems to be complex to set up artifactory as a server/service on the new machine, so we will try and put the required artifactory elements directly into the build tree.

simonkelly1973, 2019-10-23:
2e857cc does a few different things.

1.  In the main build.gradle file, it comments out all references to repositories on build.openice.info .  That guarantees that anything that can only be found on there will now no longer be found, because the build process doesn't know about those repositories any more.
2.  Using a brand new user account that has never build openice before , and killing the $HOME/.gradle directory in it then guarantees that you need to download all dependencies for an openice build.
3.  Running that build repeatedly and getting the dependencies out of the artifactory as you go along then guarantees that you have them all locally.
4.  All "new" dependencies are now in the "artifacts" directory, and included in this commit.

Of course, there may be more things that are required as we do different steps.  But this commit should at least allow a clean build from scratch without build.openice.info .

mattiasclaesson, 2020-01-08:
Its seems that the complete release and snapshot builds no longer exist on build.openice.info and that breaks our application build. This is not good practice to retire a artifactory server like this.

Why is it not possible to migrate the artifactory installation from the old to new server?