mdpnp/mdpnp #29: Talking to a device on MD-PNP

RitvikAgarwal, 2020-03-02:
Hello, 
I'm trying to read the data stream (what is displayed in the displayed graph) from a simulated device on MD-PNP GUI using a Java program. I tried executing the "hello-openice" repository (as suggested on the official page) but I'm not able to compile it from source and hence can't tweak it and check it out.

I'm feeling stuck. I'd really appreciate any guidance about this.

Thanks and Regards,
Ritvik

simonkelly1973, 2020-03-02:
Hello Ritvik

When you say "the data stream", what exactly do you mean?

I will have a look at the "hello-openice" repository now and see if that code is up to date.

simonkelly1973, 2020-03-02:
I can confirm that the hello-openice repository is not up to date - it does not compile at the moment.  We will try and review this for you.

The full openice project does build correctly with Java 1.8 at the moment.  This does include all current code - so will show everything you want to know.  On the negative side, it can be a bit more difficult to know how to find what you want in there.

If you'd like to explain a bit more about what you are trying to achieve, we will be able to help you with that.

Thanks
Simon

RitvikAgarwal, 2020-03-02:
Thanks for the prompt reply.

On launch, a small window for OpenICE Launch Settings appears (
[image]
)
The rotating graph(as shown in the attached notexact_graph.jpg) for a simulated device (as selected in open_launch2.png) must have been generated by a data stream (I just think so because a real patient will also be generating such a data stream).
[image]
[image]

P.S.: I would have attached real screenshots but I set up OpenICE on my Ubuntu machine but to check hello-openice, I switched to Windows.

simonkelly1973, 2020-03-02:
Hi Ritvik

There are a few different ways that the data is handled in OpenICE from a "producer/consumer" point of view, or a "publisher/subscriber" point of view - depending at which layer of the architecture you are looking.

At the very lowest level, all devices publish records using DDS - this is true whether they are Simulated, Network, or Serial devices.  Because DDS is a network broadcast protocol, this means that the computer that is physically connected to the device (for instance, using a serial cable) can run the "ICE Device Interface", and broadcast the device data to a completely different computer that is running the "ICE Supervisor".

This means that all data that is transferred between ICE Devices and an ICE Supervisor is transferred using DDS records - even if the ICE Device Interface is created from inside the ICE Supervisor and therefore the supervisor and the device interface are all running on the same machine (and indeed, the same process).

The supervisor also acts as the host framework for ICE Applications.  The applications are all the things on the left hand side of the main window, such as "Chart" and "Infusion Safety".  These applications can receive data via the Java event model, without knowing about DDS.  That is because the supervisor takes care of decoding DDS transmissions (that can be coming from completely different machines) and re-transmitting that data "inside" the supervisor to the applications that it is hosting.  That "re-transmission" is done using using either "Google Event Bus" or Java's "Observable Properties" - depending on exactly what you are trying to achieve.

You are quite correct that the simulated devices produce the same sort of graphs that real devices would produce.  The graphs in simulated devices are basically produced from a long array of hard coded coefficients that are then translated into array values (and then published to DDS).  For real (serial/network) devices, they are obtained by using the APIs for the specific devices.

We publish (and consume) two different types of records - Numeric records, and Array records.  Numeric records consist of a single discrete value at a point in time - for instance

systolic blood pressure = 100

Array records consist of a list of interrelated values, and are typically used to publish a chunk of a waveform (for instance, the graph for an ECG).

So, if you want to "read the stream" for a waveform, we probably need to know a bit more about exactly what you mean - but hopefully what I've written above helps us to frame the discussion a bit more accurately.

RitvikAgarwal, 2020-03-02:
Thanks. This helps a lot. 
I guess, I need to dig into the code and try to find the implementation of the concepts you've explained.
I'll get back after getting a deeper understanding of the source code. Any resources to help with that?