mdpnp/mdpnp #30: Can't find the "ice.NumericDataReader", "ice.SampleArrayDataReader"

RitvikAgarwal, 2020-03-04:
Hello,
I can't seem to find the code or jar files for the imports 
 - "**ice.NumericDataReader**", "**ice.SampleArrayDataReader**" in DevicePanel.java
in the "mdpnp-1.3.0/**interop-lab**/demo-apps/src/main/java/org/mdpnp/apps/**safetylockapplication/**"

Kindly point me to the right location.

Thanks, 
Ritvik

simonkelly1973, 2020-03-04:
Hello Ritvik

These files are auto generated from the file

ice.idl

If you have downloaded the source, running a full build will cause the .java (and .class) files to be generated.

Simon

RitvikAgarwal, 2020-03-04:
Okay, where would those generated files be located?

simonkelly1973, 2020-03-04:
Hi Ritvik

They are part of the x73-idl-rti-dds project.  In a full source download of OpenICE from the mdpnp/mdpnp repository, then after running a full build you would find the files at

data-types/x73-idl-rti-dds/build/generated/src/main/java/ice/NumericDataReader.java

All other generated files in the ice package are in the same location.  Under src/main/java, it follows the normal java package naming structure.

RitvikAgarwal, 2020-03-04:
Hi! The project is not building with Gradle 5.0 (which is written in the build.gradle file)
[image]

simonkelly1973, 2020-03-04:
Hello Ritvik

For compilation, you need a full JDK - is that Java a JRE?

RitvikAgarwal, 2020-03-04:
Output of "java -version":

openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)

That should work right?

simonkelly1973, 2020-03-04:
Hi Ritvik

As it says in the output, you need ant-javafx.jar .  That can be obtained stand alone from a JavaFX installation.  I see you are back on your linux environment.  If you do

locate ant-javafx.jar

do you have a copy of the file anywhere else?

simonkelly1973, 2020-03-04:
I'm just checking if you can get round this problem by choosing an explicit target that will not execute the tasks that have this dependency.

simonkelly1973, 2020-03-04:
Hello again Ritvik

Does your Java installation include

jfxrt.jar

or not?  You will need that, even if we got rid of the dependency on ant-javafx.jar, because it contains all the UI controls.

RitvikAgarwal, 2020-03-04:
Okay, looking for it.

RitvikAgarwal, 2020-03-04:
I was able to find jfxrt.jar (from https://github.com/qweasdzxcpoi/JavaFX/blob/master/jfxrt.jar)
But not the ant-javafx.jar
Can I compile that one from somewhere?
Also, is it possible that I could get my hands on the compiled classes in ice.<class-names>?

Thanks

simonkelly1973, 2020-03-04:
[ice-dds.zip](https://github.com/mdpnp/mdpnp/files/4287014/ice-dds.zip)

Hello Ritvik

I've attached a zip file with the java files in .  Hopefully you can use that to compile instead.

RitvikAgarwal, 2020-03-04:
Thank you so much Sir.

RitvikAgarwal, 2020-03-04:
Also, there must be mant more packages/files, generated after build.
How can that problem we solved?

sotirislilio, 2022-11-28:
> Hi! The project is not building with Gradle 5.0 (which is written in the build.gradle file)
> [image]
> 

 I am facing a similar issue with JavaFX.. @RitvikAgarwal did you build it successfully? If so, could you please share the steps you followed?