mdpnp/mdpnp #37: Update device configuration remotely ?

synyster0fa7x, 2020-03-24:
Hey !

I am part of a team participating in a French healthtech hackathon which is taking place right now, working on several projects specific to the fight against the Covid-19, and the improvement of the working conditions of healthcare staff.
We are looking for a way to interact remotely with various medical devices, mainly respirators and electric syringe pumps.

This repository seems to suit our needs in terms of information gathering. I noticed some commits allowing perhaps, in the future, to configure some devices remotely (provided that the manufacturer authorizes this action).

What opinion do you have on this?
Do you have any roadmap with this last point?

In advance, thank you for your answer.

simonkelly1973, 2020-03-24:
Hello

Certainly, it is possible to use OpenICE to remotely control or configure devices, if there is a publicised API for the device.  OpenICE can interact with devices over network or serial connections.

I'm sure we would be happy to do some work with you on this.

synyster0fa7x, 2020-03-25:
Thank you for this quick response !

We are currently contacting some manufacturers in order to try to obtain documentation and perhaps hardware allowing us to implement a technical solution with the best conditions.

Is there a way to contact you other than the tickets, so as not to crowd them ?

simonkelly1973, 2020-03-25:
Hi - I emailed you directly to two email addresses I saw in your commit history earlier on today.  Let me know if you didn't receive anything.