mdpnp/mdpnp #43: Support OpenEMR integration for patient list

simonkelly1973, 2020-06-11:
As part of the OpenEMR integration, one of the requirements is to take the list of patients from OpenEMR.  OpenEMR has a suitable API to provide the patient list after authenticating.

simonkelly1973, 2020-06-11:
032f237 does an initial implementation.  OpenICE has always supported getting a patient list from an external source, and the central class for that is EMRFacade .  Previously, that supported getting access to patient data from a FHIR server, if the FHIR server location was passed in.  The "instance" of EMRFacade that is used for any session (a subclass of EMRFacade) is chosen in the Factory class inside EMRFacade.  For the purposes of this initial implementation of OpenEMR integration, we have made OpenEMR the primary source - i.e. if OpenEMR is specified, it is used.  There is no attempt to cooperate and use mulitple sources if FHIR or JDBC are specified - if OpenEMR is specified, all others are ignored.

We have the new OpenEMRImpl class that does the actual work of logging in and getting the patient list from the OpenEMR server.  Because the OpenEMR server requires authentication, we need a username and password, and also a "scope" (that identifies the instance of OpenEMR on the server that we are interested in).  To avoid cluttering up the splash screen GUI, we read those three properties from a new file, iceopenemr.properties in the user home directory.

The fetchAllPatients method in the new implementation is called by the refresh method in the parent class.  We check if we are already logged in, and if not we do so, using the API calls.  Exception handling should be reviewed.  If we fail to log in, or indeed fail to parse the patient list etc., then everything is logged to the demo-apps.log file, but there's no UI notification.

The format of patient data from OpenEMR is a simple JSON object format and the parsing is easy and done in addOpenEMRPatients.

Everything else in the change is basically "plumbing" - adding fields to the Configuration dialog to allow the server name to be entered, setting a system property to allow the value to be picked up down the line etc. etc.

haroonalisa, 2022-01-24:
OpenICE fails to get authentication token from current OpenEMR version (v6.0). Can you share the supported OpenEMR version please?

haroonalisa, 2022-01-24:
@simonkelly1973 No, There's a **icepatientdb.properties** named file though.

simonkelly1973, 2022-01-24:
That's the list of predefined or built in patients.  We'll give you our example properties file shortly.

simonkelly1973, 2022-01-24:
@haroonalisa Here's a sample iceopenemr.properties

cat /home/simon/iceopenemr.properties

username=yourusername
password=yourpassword
scope=default

The yourusername and yourpassword are just a username and password that you can use to log in to your OpenEMR instance via the normal web interface.

haroonalisa, 2022-01-25:
Thanks a lot, Was able to get an authentication token.
Although fetching patient list from OpenEMR had an issue, I think this is the problem:

https://github.com/mdpnp/mdpnp/blob/3d3f24bf12226353d18b3f969f8c20977a6997f1/interop-lab/demo-apps/src/main/java/org/mdpnp/apps/testapp/patient/OpenEMRImpl.java#L135-L139

The KEY for Date of birth in openEMR's API response is "dob" and here its trying to get value for the Key "DOB". This solution is working for me currently. Also the try-catch doesn't seem to be working