mdpnp/mdpnp #45: Broken Links for Device-Adapter Disk Images

greipfrut, 2020-06-22:
On the setup guide for Device Adapters (https://www.openice.info/docs/4_device-adapter-setup.html), the links to **Revision C Disk Image** and **Revision B Disk Image** are broken.

The **Beaglebone Disk Image Notes** (https://www.openice.info/devdocs/beagle.html) also appears to be outdated if one were to try to create their own images due to the broken links noted above.  For instance, under **Install Java** of the notes, the elinux.org link referenced (http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Oracle_Java_8) no longer appears to provide the intended documentation, which causes issues with missing **ant-javafx.jar** if the documentation isn't pinning a specific version of the JDK that would contain the jar.

simonkelly1973, 2020-06-24:
Hello.
We did not create any new physical beaglebone devices in the MDPNP lab for a long time.  Disk images that are miasing will be due to changes in web hosting.

Regarding the documentation-it is probably outdated now because the hardware would be different if starting out with new beaglebones now.

Do you have a particular use case for Beaglebone now?

ko-sinus, 2022-03-04:
Hello,
I'm reopening this issue as you recently released a new version of OpenICE. I was wondering if beaglebones are still used as Device-adapter as images and tutorial links concerning them are still broken.

ioannisgkouzionis, 2022-11-27:
Same question with @ko-sinus. I'd like to also ask if Raspberry Pi 4 can be used as device-adapter. Thank you @simonkelly1973 .

simonkelly1973, 2022-11-27:
Yes, Raspberry Pi can be used as a device interface.  Although the Raspberry Pi does not have a traditional serial port, you can simulate a serial port with the GPIO pins on the Pi.  There are lots of references for the correct pins, such as this one

https://www.abelectronics.co.uk/kb/article/1035/serial-port-setup-in-raspberry-pi-os

You can get JDK 8 for Raspberry Pi and then compile and run OpenICE using gradle.

sotirislilio, 2022-11-28:
@simonkelly1973 with JDK 8 for RPi installed I was not able to compile and run OpenICE... Indeed there are issues with jar files like the ant-javafx.jar, as @greipfrut said.
I ve also used BellSoft Liberica JDK which includes javafx but build fails too.

wodin, 2024-06-10:
@simonkelly1973 Are there any notes on how the old images were built so that someone can have a go at getting it working on a newer device? I see the Rev C image filename contains "lxde", which suggests it requires X11? That it surprising to me. I would have expected only the supervisor to require a GUI?

What parts of OpenICE constitute the "device-adapter"? I see the init script runs `/home/debian/OpenICE/OpenICE.current/bin/OpenICE`, so does each beaglebone (or whatever) run basically the same software (GUI and all) as the supervisor?

Thanks

simonkelly1973, 2024-07-10:
@wodin what configuration are you looking to run the device adapter in? Completely headless with no graphics where you just want to connect a single device?  What "newer" model of BeagleBone are you interested in?  As you may have seen, we have a fully working version of OpenICE on Raspberry Pi Model 4 now, with minimal adaptations to the code.  So we expect it to be easier to work with on BeagleBone now as well.

wodin, 2024-07-12:
Hi @simonkelly1973 
I haven't decided yet on what device to use. But yes, the plan would be for it to be headless and probably to connect a single device. But I'm not really sure what makes the most sense.
I hadn't seen that you have it working on a Raspberry Pi model 4. I see now that Raspberry Pi support is mentioned in the latest release notes. Is the image for RPI model 4 available somewhere? Or instructions on how to build it? Does it currently require a GUI?

simonkelly1973, 2024-07-15:
Hello @wodin 

We didn't produce an image for the Raspberry Pi ourselves - we just got it working with the model 4.  We don't have any particular special instructions for it, because as far as we can tell, the Pi doesn't require any special instructions any more.  You can just download JDK 17, and then compile and run OpenICE without doing anything specific to the Pi.

We added the required native libraries into the source code on GitHub and adapted the gradle build script so that it detects the ARM platform and uses those new libraries.

In the meantime, I have acquired a new BeagleBone Black Rev C, which I plan to test OpenICE on.  However, unfortunately at the moment it is unstable and reboots constantly, so I can't do anything with it.

wodin, 2024-07-15:
@simonkelly1973 OK thanks for the clarification

simonkelly1973, 2024-07-15:
> @simonkelly1973 OK thanks for the clarification

I don't suppose you have any experience of debugging this reboot issue on a BeagleBone do you ?  Out of interest, what do you power yours with?  I'm using the USB connector at the moment rather than the 5V barrel connector, but I'm thinking of buying a separate 5V/3A barrel connector PSU.

wodin, 2024-07-17:
@simonkelly1973 sorry, I have some experience with older Raspberry PIs (and with custom firmware on ADSL routers and WiFi APs). The reason I was asking about Beaglebone is because the OpenICE docs mentioned it.

But does yours reboot predictably or just at random? Do you get any sort of error info or does it just seem to die? Trying a different power source does seem like a good idea.

simonkelly1973, 2024-07-17:
> @simonkelly1973 sorry, I have some experience with older Raspberry PIs (and with custom firmware on ADSL routers and WiFi APs). The reason I was asking about Beaglebone is because the OpenICE docs [...]

I've now obtained a 5V 3A power supply with the appropriate barrel connector.  The device has been running for about 5.5 hours now, so that seems to be the issue.

There isn't enough space left on the eMMC storage to install everything, so I'm in the process of getting the latest image burned onto an SD card that will be big enough.