mdpnp/mdpnp #84: Problem HL7 Exporter

DanielaRC94, 2022-08-01:
The Hl7 exporter does not allow messages to be sent to an address and port, in the FHIR option the information is not displayed.
Could you tell me how to get the information?

simonkelly1973, 2022-08-01:
Hello @DanielaRC94 

Could you tell us a bit more about what you are trying to do?  In the HL7 Exporter application - are you saying that the Host: and Port: fields are disabled, or not visible?

In the attached jpg, that shows the normal view.

[image]

DanielaRC94, 2022-08-02:
Good morning
Simon
I am simulating a vital signs monitor in Open Ice, I select Hl7 Exporter,
in the FHIR DSTU2 section no information is displayed, in the v2.6 section
an ORU message is displayed, but only the OBX of the heart rate parameter
is displayed.
What I require is to display the information of the ORU message of the
other parameters simulated by the vital signs monitor (Pulse, SPO2, etCO2
etc)
And take this information to a server or a mirth connect channel.

[image: image.png]

[image: image.png]

simonkelly1973, 2022-08-02:
Hello Daniela

If you make the host and port fields at the bottom of the screen to be empty, and the click start, do you get more content in the main text area of the screen?  The attached screenshot here shows some messages when using the OpenICE Simulated Multiparameter Monitor.

[image]

DanielaRC94, 2022-08-02:
Thank you very much
How could I send this information to a server?

DanielaRC94, 2022-08-02:
> Hello Daniela
>
> If it works when you do not specify a host, there is probably a problem
> with communicating with the host/port that you do specify. Are you able to
> attach the log file from your user home directory? It will be called
> demo-apps.log
>
> Thanks
>
> Simon

In this part?

[image: image.png]

simonkelly1973, 2022-08-02:
Unfortunately when I look at your messages I don't see the images etc - either on this GitHub issue page or the emails I receive.  Are you using Unix or Windows?

DanielaRC94, 2022-08-02:
I am using email- windows

simonkelly1973, 2022-08-02:
Your replies just look like this

[image]

DanielaRC94, 2022-08-02:
[image]

DanielaRC94, 2022-08-02:
[image]
[image]

DanielaRC94, 2022-08-02:
Can you see the images?

simonkelly1973, 2022-08-02:
Hello.

When you start the application and the splash screen appears, then the "FHIR Server" there is not to do with the HL7 export app.  That text box is used to specify an HL7 server that is then used as a source of patient information.  Those patients are visible in the main window of OpenICE in the border at the bottom, in the popup box labelled "Select a patient:".

The HL7 Exporter window has a different server specification - using the "Host:" and "Port:" fields at the bottom of the screen.

In the picture we can now see, you have the messages in the main part of the window.  You should be able to specify the host/port details for your HL7 server in there.