mdpnp/mdpnp #88: Problem running on Raspberry Pi 4

sotirislilio, 2022-12-01:
I am trying to run OpenICE 1.5 on a Rpi 4b 2gb. However, when running the OpenICE script errors related with javafx appear. I 've also tried to manually add the javafx files but new issues appear. For instance:
"Graphics Device initialization failed for :  es2, sw"
"Error initializing QuantumRenderer: no suitable pipeline found"

Moreover, errors appear even when I try to build the project with gradle. For example ant-javafx is missing:
"Couldn't find Ant-JavaFX-library, please make sure you've installed some JDK which includes JavaFX (e.g. OracleJDK or OpenJDK and OpenJFX), and JAVA_HOME is set properly."

JDK 8 is installed as in the instructions. I've also checked if JAVA_HOME has been set properly.

Any help would be much appreciated.

simonkelly1973, 2022-12-01:
Hello

The first issue seems to be related to missing native libraries as part of the Java environment.

The second issue is also related to the installed version/vendor of Java.

Exactly which version do you have installed, from which vendor?  The only one that we know works reliably is the official Oracle one for JDK 8.

sotirislilio, 2022-12-01:
@simonkelly1973 I ve installed Oracle's JDK 8u331. I ve also tested it with Open JDK 8 in another rpi 4 without success...

sotirislilio, 2022-12-01:
Regarding the first issue, is there a way to identify which libraries are missing?

simonkelly1973, 2022-12-01:
> @simonkelly1973 I ve installed Oracle's JDK 8u331. I ve also tested it with Open JDK 8 in another rpi 4 without success...

Although I'm not in front of my usual Linux machine at the moment, there is the following on my Windows machine

C:\Program Files\Java\jdk1.8.0_202\lib>dir ant-javafx.jar
 Volume in drive C is Windows
 Volume Serial Number is BE0D-CFB5

 Directory of C:\Program Files\Java\jdk1.8.0_202\lib

29/03/2019  10:57         1,500,582 ant-javafx.jar
               1 File(s)      1,500,582 bytes
               0 Dir(s)  659,922,804,736 bytes free

C:\Program Files\Java\jdk1.8.0_202\lib>

Do you have the equivalent named file somewhere under your JDK installation?

sotirislilio, 2022-12-01:
Unfortunately, there is no such file in the jvm directory. I've also tried to search and download it from the web and manually add it to the jvm dir.  However, I wasn't able to find it...

simonkelly1973, 2022-12-01:
That might suggest that JavaFX support is not available as part of Java for this platform.  We'll try and look at it more but ultimately if it's a platform constraint then it won't be easy for us to do anything about it.

sotirislilio, 2022-12-01:
Ι also have a beaglebone revision C. Are there any chances to face the same issues there too? Is it tested there? and if yes could please you share the steps to run it there?

Thanks @simonkelly1973 .

ioannisgkouzionis, 2022-12-03:
@simonkelly1973 I tried installing the latest version of OpenICE on a Beaglebone Black Rev C with jdk8u351 (that's the version you have in the link https://www.openice.info/devdocs/build.html) and I got the same errors as @sotirislilio. I also attach the reference images. Could you please assist on that as following the exact steps mentioned in the OpenICE documentation it's impossible to make it work on both BB and Rpi? Thank you very much in advance.
[image]
[image]

simonkelly1973, 2022-12-03:
Hello.

Yes, you experience the same problem, for the same reason.  The basic problem is this

1.  OpenICE relies heavily on JavaFX for the UI components.
2. JavaFX in version 8 of Java was included in the JDK for x86, x86_64 and similar "common" architectures.
3. On platforms where JavaFX is not available as part of the JDK or JRE, OpenICE cannot run.

We plan to tackle this problem in the next month or two by

1.  Upgrading OpenICE to use the latest stable version of Java.  We may opt for 17 for Long Term Support (LTS), or we may use 19.
2.  Because JavaFX is not a core part of the Java JRE/JDK any more (it was removed in Java 9), we will be using OpenJFX.  This seems to have better support for ARM platforms, so after that is done we will hopefully be able to get back to prebuilt images for things like RaspberryPI and BeagleBone.