mdpnp/mdpnp #95: Running OpenICE in Raspberry Pi 4

kallolkk, 2023-09-14:
Hi All,
I want to run OpenICE in Raspberry Pi 4. So far I have tried various approaches. I will log them below:
1. Running OpenICE a Raspbian (x86) VM: This approach works with JDK8 with inbuilt JavaFX (from oracle)
2. Running OpenICE in Raspberry PI 4 hardware: I have tried couple of options
    a)  JDK 8 (from oracle) for aarch64 does not have the JavaFX and hence it does not work.
    b) Someone mentioned Rasbian 2015 comes with preinstalled JDK8 with JavaFX. Its true, so I extracted the JDK8 and tried to run in recent Raspbian (32bit) with Raspberry Pi4. JDK 8 works but it fails to start OpenICE. It shows page memory error.  The extracted JDK8 does not work in Raspbian 64 bit. 
    c) I have tried OpenJDK from Azul and it did not worked.
I want to know how i can run OpenICE in Raspberry PI 4. Also, is it possible to compile the OpenICE code repo using recent JDK (in backward compatibility mode), so that it works with the recent version.

Thanking You All.

Best Regards
Kallol

simonkelly1973, 2023-09-14:
Hello Kallol

More recent versions are probably better for running as Raspberry Pi.  We have been thinking about improving support for the RPi.  If you check out the branch

upgrades22

that branch is designed to compile using Java 17, using OpenJDK, and using OpenJFX for the JavaFX libraries.

Do try that branch out and let us know how it goes.

Thanks

Simon

kallolkk, 2023-09-18:
Hi Simon,
Thank you for pointing me to _upgrades22_ branch. I did the following: 
1. I have installed OpenJDK 17.0.8 with fx from Liberica. I have used SDKMAN (https://github.com/sdkman/sdkman-cli) for this installation. My Raspbian is the latest 64bit edition running on RPi 4. 

2. I cloned the upgrades22 branch of the mdpnp repo and run the _buildanddeploy.sh_ script. After the build phase it was looking to collect to port 22 for a service which it never received and i had to stop it. Please see the command log. There are 4 warnings and 1 error. However the build was successful. 
[bandd.txt](https://github.com/mdpnp/mdpnp/files/12644151/bandd.txt)

3. After that, I thought of running _./gradlew :interop-lab:demo-apps:run_ only. This time the OpenICE window came (image1) and when i clicked the Strat ICE_Supervisor button, it showed an error (Image 2)
[image]
[image]
[gralewrun.txt](https://github.com/mdpnp/mdpnp/files/12644177/gralewrun.txt)

Can you please let me know, what it wrong in my setup?

Thanking you. 
Best regards
Kallol

simonkelly1973, 2023-09-25:
Hello Kallol

Just to let you know that we have just acquired a new Raspberry PI 4 and we are looking at building this.

You mentioned that compilation worked correctly.  But for us it looks like the JavaFX plugin doesn't work.  It seems you have a different version of the JDK.  We will try that.

The final issue you seem to encounter is that we only detect that the OS is Linux and we try and load the shared objects for RTI DDS for Linux - but these are x64 and we need the aarch64 objects instead.  These are available - we just need to switch to loading them.

We'll come back to you soon.

simonkelly1973, 2023-09-26:
Hello @kallolkk - we have updated the upgrades22 branch to support Raspberry PI when using 64 bit userland.

Just export JAVA_HOME to point to the correct JDK installation and then do

./gradlew run

And you should be up and running.  Do let us know how you get on!

kallolkk, 2023-09-27:
Hi Simon,
Thank you. Can you please share which JDK you are using. I am still unable to start the ICE_Supervisor screen. 
I am using the same OpenJDK 17.0.8 with fx from Liberica. 

Thanking You. 
Best Regards
kallol

simonkelly1973, 2023-09-27:
Hello Kallol

simon@raspberrypi:~ $ which java
/home/simon/.sdkman/candidates/java/current/bin/java
simon@raspberrypi:~ $ java -version
openjdk version "17.0.8.1" 2023-08-24
OpenJDK Runtime Environment Temurin-17.0.8.1+1 (build 17.0.8.1+1)
OpenJDK 64-Bit Server VM Temurin-17.0.8.1+1 (build 17.0.8.1+1, mixed mode, sharing)

We actually "chose" that version by following your original suggestion of using sdkman, and just did

sdk install java

we get the following from "sdk current java"

simon@raspberrypi:~ $ sdk current java

Using java version 17.0.8.1-tem

Hopefully that can help you to get exactly the same one as us.  In the meantime, do you have a file

demo-apps.log

in the user home directory?

I'm sure we are close to having it working with you.

Thanks!

Simon

kallolkk, 2023-09-29:
Hi Simon,
I have tried with the exact version of JDK and still its not showing the supervisor menu for me. Please see check my demo-apps.log. Sorry for disturbing again and again. 
[demo-apps.log](https://github.com/mdpnp/mdpnp/files/12756472/demo-apps.log)
Kindly provide me some suggestion to fix this issue. 
Thanking You
Best Regards
Kallol

simonkelly1973, 2023-09-29:
Hello

Don't worry about being in touch-we are really interested to get this working with you.

There are some exceptions in there but they don't look like they should prevent you from starting. Just to clarify

Do you see the splash screen and do "start" in there and then see nothing after the splash screen disappears?

We'll compare it with a demo-apps.log we get here.

kallolkk, 2023-10-03:
Hi Simon, 
My RPI had some hardware issues. Today, I purchased a new RPI4 (8GB) and performed the steps. It worked perfectly. Thank you very much for helping me. 

Best Regards
Kallol 
   
[image]

simonkelly1973, 2023-10-03:
Hello Kallol.

That's really excellent to hear, thanks!

Do keep in touch with any other questions or suggestions.

Now it's working, could we ask what your interest in it is?  Are you doing research, getting data from a real device, or just interested in the product architecture?

Thanks and regards

Simon

kallolkk, 2023-10-03:
Hi Simon, 
We are doing research. We are researching on security of healthcare products. We are focusing or the communication side of the OpenICE. 

Thanking You 
Best regards
Kallol

kallolkk, 2023-10-12:
Hi Simon,
As part of my research work, i will be installing OpenICE in RockPro64 as well. Those SoCs will be using Ubuntu Mate and Armbian. So, I thought of installing OpenICE in Ubuntu Mate Release 22.04 LTS (Jammy Jellyfish) 64-bit (Kernel Linux 5.15.0-1011-raspi aarch64). I followed the same steps which worked for Raspbian OS. However, with Ubuntu Mate 22.04 it does not work. Please see the details below. 
1. I received the following error with upgrades22 branch with java Temurin-17.0.8.1. I did a little googling with the error message. Some blog said its due to upgraded GCC version. The previous raspbian -bullseye has gcc-10.04 and this ubuntu mate has gcc -11.2.
[image]

2. I tried the rti_dds_7 branch as well, which compiles and brings the first landing page ok. But fails to start the ICE supervisor window (the old problem). 
 
[image]

Can you please have a look into it. 

Thanking You
Kallol

simonkelly1973, 2023-10-12:
Hello Kallol

As you have seen, we do have a few branches that we are using at the moment, for different things.  Like all good branches, at some point they will all merge back into master, but for now they are separate.

You are correct about the compiler version being the issue with the data-types project.  What happens with that project is that the .idl files are translated into Java code.  Because the idl files have dependencies between them, they use compiler directives like #include.  Those directives are resolved by using the C pre-processor.  That fails on the latest GCC versions, but that failure is fixed in the latest RTI versions.  The latest rti version is in the rti_dds_7 branch, and that's why that compiles.

Now when we did the other fixes for Raspberry Pi recently, the actual fix was to add a check for architecture rather than just platform, and to add in the aarch directory under interop-lab/demo-apps/native/libs.  That provides "ARM" versions of the libraries.  Those changes are missing in the rti_dds_7 branch.

So as a quick test, you could try copying the

aarch directory from interop-lab/demo-apps/native/libs
and
build.gradle file from interop-lab/demo-apps

from a checkout of upgrades22 into a checkout of rti_dds_7, and that should hopefully get you working.

Let me know if that doesn't make sense!  Or if it works.

Thanks

Simon

kallolkk, 2023-10-13:
Hi Simon, 
Thank you for the suggestion. I did the same. I received a different error while trying to start the ICE Supervisor. Please see below. 
[image]

Thanking you 
Best Regards
Kallol

kallolkk, 2023-10-20:
Hi Simon, 
Can you please suggest me how I can overcome the above error. Sorry to disturb you again and again. 

Thanking You
Best regards
Kallol

kallolkk, 2023-11-23:
Hi Simon,
Can you please suggest how i can resolve the issue with latest GCC version and ARM64 units.
Thanking You 
Best Regards
Kallol

kallolkk, 2023-11-29:
Hi Simon, 
I know the cause of the last error in rti_dds_7 branch. It is due to a miss match between versions. 
The aarch directory file from interop-lab/demo-apps/native/libs of upgrades22 branch is compatible with DDS6.2 version not DDS7. It is causing conflict with rti_dds_7/artifacts/nddsjava.jar (which is from DDS7) file. That is why i am getting that error. If you happen to have aarch files for DDS7. Can you please share them. I am unable to get hold of the files. It will help others who are trying to run OpenICE with GCC 11+ and ARM platform. Thank you for your help.

Best regards
Kallol

simonkelly1973, 2023-12-01:
Hello Kallol

Sorry about the long long long silences on this.  We have now updated the rti_dds_7 branch so that it uses RTIs latest 7.2 release, and has the matching binaries for aarch 8 for Raspberry PI.  The build.gradle has the extra check for architecture as well as OS.

We had to make hacks to the code for now.  In this release it seems that RTI have changed the data type for "seconds" in their timestamp from an int to a long, so for now we have just added casts from long back to int.  This can only be considered a quick and dirty fix but we'll need to change a lot more of the OpenICE code to handle longs with our timestamp code.

Anyway, right now the

rti_dds_7

compiles and runs cleanly on RaspberryPI.  Let us know how it goes, and sorry again when it looked like we were ignoring you!

Regards

Simon

kallolkk, 2023-12-05:
Hi Simon,
Thank you very much. I have just tested OpenICE rpi_dds_7 branch on Rock Pi 64 running Armbian. It works flawlessly. I have also tried in Debian VM running on MAC (M2) with Parallels. OpenICE works with no issues.

Thanking You 
Best Regards
kallol