mdpnp/mdpnp #101: is this software support  philips IntelliVue MP60?

pipiljf, 2024-05-11:
Hi
I dont know if  this software support  philips IntelliVue MP60. I had never success when I try to connect with MP60. 
With infomation as follow picture 
[image]

Your help will very much appreciated！

simonkelly1973, 2024-05-11:
Hello.

Can you confirm if you are using a serial connection?

If so, what serial port parameters are you using?

SiergiejPOL, 2024-05-11:
i can confirm that intellivue mp50, using ethernet cable with xds connection 
but software is detect those monitor as mp70

pipiljf, 2024-05-12:
> Hello.
> 
> Can you confirm if you are using a serial connection?
> 
> If so, what serial port parameters are you using?

Hello,simonkelly1973.Thanks for your reply .

I am indeed using a serial connection,  like below picture
[image]
 And I just use default parameters to connect because I didnot find where to set serial port parameters in OpenICE. 

Do you know what happened？Thank you very much!

pipiljf, 2024-05-12:
> i can confirm that intellivue mp50, using ethernet cable with xds connection but software is detect those monitor as mp70

Hi SiergiejPOL ,thanks for your reply.

Could you tell me how you did that? 

I follow tutorial on official website(https://www.openice.info/docs/6_example-setups.html),and run OpenICE Supervisor and Device-adapter in same one Windows  laptap, but it donot work.

SiergiejPOL, 2024-05-12:
Hey [pipiljf]

Your monitor MP60 have RJ45 / Ethernet port, if yes just connect using ethernet casble to your network, next download and install  IntelliVue XDS Application [http://intellivue.medical.philips.com/xds/]  (i am using for getting ip address from different range than my router and using BOOTIP for this, also ato print to pdf reports from monitor)

if u connect monitor to your network and with success install and configure XDS software your monitor will get ip address from range of your network. at last open OpenICE choice Supervisor, click on button "Create an ICE device Adapter" choice not serial but network... place ip adress of your intellivue mp60 ... and here you go data should be transfered and showed on application