mdpnp/mdpnp #110: Help with Compiling Project Without GUI and Automating Data Recording

jithin-c-s, 2024-09-11:
I'm looking for help with compiling project from the command line without GUI dependencies. Additionally, I need to automate data recording when my PC connects to a Philips MP70 monitor. Any advice on build configurations or scripts for these tasks would be appreciated.

Thanks!

simonkelly1973, 2024-09-12:
Hello

Could you tell us which BeagleBone board you are using?

We recently investigated the modern Black variant but it's a 32 bit board and that it is not good for Java support.

What type of automation are you thinking of?  Thr data exporter application is a UI based app-that is necessary to gather options like the destination file and format like CSV and so on.

It might be possible to create a non-ui wrapper around that but we'd need to understand more about what you wanted to achieve.

Thanks

Simon

jithin-c-s, 2024-09-12:
If the BeagleBone's 32-bit compatibility is an issue, are there alternative options I can consider, such as the Raspberry Pi?

simonkelly1973, 2024-09-12:
Hello.

Firstly, yes, there is excellent support for the Raspberry Pi.  On a Pi version 4 for example, OpenICE can be cloned from GitHub and will compile and run without any special steps.

There are two major components of OpenICE.  One if the "device interfaces" - there are implementations of the protocols for the different devices such as the Philips Intellivue.  To connect to a device and have data published from it, you can run an instance of a device interface completely headless.  This is already supported.

The second major component is the applications.  These do things like closed loop control (adjusting the speed of a pump in response to blood pressure changes), or data export.  The data export tool is designed to write to a database or a CSV file.  All applications are GUI based though.

However, a major innovation in the current release of OpenICE is improved integration with Python.  As you may know, when devices are connected to OpenICE using the specific device interface, they publish data using the DDS (Data Distribution Service) protocol.  This is a network protocol with formatted data records.

In older versions of the product it was difficult to consume this data outside of OpenICE by "subscribing" to what OpenICE "published".  In the latest version of OpenICE, we have upgraded to use the latest release of DDS by RTI (https://rti.com) .  This version has greatly improved and simplified integration with Python, making it very easy to consume the data that devices are publishing using a Python script.

Of course your Python script can be headless, and do whatever you want with the received data.  So it should be quite straighforward to fulfill your requirements.  If you look in an up to date copy of the source, you will find

[Python examples](https://github.com/mdpnp/mdpnp/tree/master/data-types/x73-idl/src/main/idl/ice/numeric_subscriber)

that shows how to subscribe to the "numeric" data type.  Numerics are single data points such as pump speed or a heart rate.  Subscribing to other data types such as the waveform datatype is certainly possible as well.

Hopefully this helps you to get going.  Do keep in touch with other questions or progress reports.

jithin-c-s, 2024-09-12:
Thanks for the information. It looks like everything I need, especially the Python integration and headless device interfaces, is already available. I’ll get the Raspberry Pi set up, and if I face any challenges, I may need some detailed guidance to help with the process.

I’ll reach out once it's up and running!

jithin-c-s, 2024-09-12:
>> **The second major component is the applications. These do things like closed loop control (adjusting the speed of a pump in response to blood pressure changes), or data export. The data export tool is designed to write to a database or a CSV file. All applications are GUI based though.**

I hope there's a workaround to bypass the GUI part, though.

jithin-c-s, 2024-09-12:
Using a separate system for the GUI might be necessary, but that may not be feasible in my case.

simonkelly1973, 2024-09-12:
If you want to run a device interface,that's completely headless.  You need the relevant JAR files just so the code as a whole compiles, but you don't need a graphics display during compilation, or during runtime just to run the device.  Running a device interface fully headless works fine.

And if you are just going to subscribe to the data using Python, the Python code is yours to execute as a separate process from OpenICE - so that can be headless too.

jithin-c-s, 2024-09-12:
oh that's great !! i will update once the set up and running.

jithin-c-s, 2024-09-13:
Hi,

I’ve successfully set up the Raspberry Pi. Below are the details of the board:

board		: Raspberry Pi 4 B (8 GB version)
os		: Raspberry Pi OS Lite(64 bit)
cloned		: https://github.com/mdpnp/mdpnp/releases/tag/v2.0.0

installed java 17 and build-essential
executed ./buildAndDeploy.sh 

but faced some issue 

**hmicro@hmicro-pie:~/jithin/mdpnp $ sh ./buildAndDeploy.sh 
Downloading https://services.gradle.org/distributions/gradle-7.6-bin.zip
...........10%............20%...........30%............40%............50%...........60%............70%............80%...........90%............100%

Welcome to Gradle 7.6!

Here are the highlights of this release:
 - Added support for Java 19.
 - Introduced `--rerun` flag for individual task rerun.
 - Improved dependency block for test suites to be strongly typed.
 - Added a pluggable system for Java toolchains provisioning.

For more details see https://docs.gradle.org/7.6/release-notes.html

Starting a Gradle Daemon (subsequent builds will be faster)
NOTE: To build with OpenSplice compatibility please set the OSPL_HOME and SPLICE_TARGET environment variables.

> Configure project :data-types:x73-idl-rti-dds
rtiddsgenExplodeResources started
rtiddsgenExplodeResources ended
<=====--------> 44% CONFIGURING [52s]

> Configure project :interop-lab:demo-apps
Project :interop-lab:demo-apps => no module-info.java found

FAILURE: Build failed with an exception.

* Where:
Build file '/home/hmicro/jithin/mdpnp/interop-lab/demo-apps/build.gradle' line: 328

* What went wrong:
A problem occurred evaluating project ':interop-lab:demo-apps'.
> Could not resolve all files for configuration ':interop-lab:demo-apps:runtimeClasspath'.
   > Could not find javafx-fxml-17.0.11-linux-aarch64.jar (org.openjfx:javafx-fxml:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-fxml/17.0.11/javafx-fxml-17.0.11-linux-aarch64.jar
   > Could not find javafx-web-17.0.11-linux-aarch64.jar (org.openjfx:javafx-web:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-web/17.0.11/javafx-web-17.0.11-linux-aarch64.jar
   > Could not find javafx-controls-17.0.11-linux-aarch64.jar (org.openjfx:javafx-controls:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-controls/17.0.11/javafx-controls-17.0.11-linux-aarch64.jar
   > Could not find javafx-swing-17.0.11-linux-aarch64.jar (org.openjfx:javafx-swing:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-swing/17.0.11/javafx-swing-17.0.11-linux-aarch64.jar
   > Could not find javafx-media-17.0.11-linux-aarch64.jar (org.openjfx:javafx-media:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-media/17.0.11/javafx-media-17.0.11-linux-aarch64.jar
   > Could not find javafx-graphics-17.0.11-linux-aarch64.jar (org.openjfx:javafx-graphics:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-graphics/17.0.11/javafx-graphics-17.0.11-linux-aarch64.jar
   > Could not find javafx-base-17.0.11-linux-aarch64.jar (org.openjfx:javafx-base:17.0.11).
     Searched in the following locations:
         https://repo.maven.apache.org/maven2/org/openjfx/javafx-base/17.0.11/javafx-base-17.0.11-linux-aarch64.jar

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 7m 52s
ls: cannot access 'interop-lab/demo-apps/build/distributions/OpenICE-*.zip': No such file or directory
Shipping OpenICE- bundle to 10.0.1.200
ssh: connect to host 10.0.1.200 port 22: Connection timed out
Awaiting shipment completion
ssh: connect to host 10.0.1.200 port 22: Connection timed out
scp: Connection closed
Decompressing to 10.0.1.200
Awaiting decompression
ssh: connect to host 10.0.1.200 port 22: Connection timed out
Stopping device-adapter on 10.0.1.200
ssh: connect to host 10.0.1.200 port 22: Connection timed out
ssh: connect to host 10.0.1.200 port 22: Connection timed out
Copying more files to 10.0.1.200
ssh: connect to host 10.0.1.200 port 22: Connection timed out
scp: Connection closed
ssh: connect to host 10.0.1.200 port 22: Connection timed out
scp: Connection closed
ssh: connect to host 10.0.1.200 port 22: Connection timed out
ssh: connect to host 10.0.1.200 port 22: Connection timed out
Restarting device-adapter on 10.0.1.200
ssh: connect to host 10.0.1.200 port 22: Connection timed out**

simonkelly1973, 2024-09-13:
Hi.

This is something we have only noticed very recently.  There is a workaround.

2.  Edit the following two files

 

./interop-lab/demo-guis-javafx/build.gradle
./interop-lab/demo-apps/build.gradle

 

3.  Both of those contain a line

version = "17.0.11"

Just edit that to

version = "17.0.8"

After that you should be good to go.  Please let us know.

jithin-c-s, 2024-09-13:
with the above changes build is sucessfull

**BUILD SUCCESSFUL in 3m 15s
52 actionable tasks: 52 executed
Shipping OpenICE-1.5.0-SNAPSHOT bundle to 10.0.1.200
ssh: connect to host 10.0.1.200 port 22: Connection timed out
Awaiting shipment completion**

attaching the full console output
[console_out.txt](https://github.com/user-attachments/files/16996336/console_out.txt)

simonkelly1973, 2024-09-13:
This all looks fine.  The "shipment" output is very old.  As you can probably see, it's designed to deploy the resulting build to a different machine, unzip it on that box, and then start the service on that machine.

That's only ever going to work on the internal network.  To just compile and run the code yourself you can just do

./gradlew run

or to just compile without running

./gradlew compileJava

Anyway - you should be fine to carry on from here.

jithin-c-s, 2024-09-13:
Since I'm new to Java, I'm having trouble understanding and resolving the error.

hmicro@hmicro-pie:~/jithin/mdpnp $ ./gradlew run
NOTE: To build with OpenSplice compatibility please set the OSPL_HOME and SPLICE_TARGET environment variables.

> Configure project :data-types:x73-idl-rti-dds
rtiddsgenExplodeResources started
rtiddsgenExplodeResources ended

> Configure project :interop-lab:demo-apps
Project :interop-lab:demo-apps => no module-info.java found

> Task :interop-lab:demo-apps:run
Exception in thread "main" java.lang.UnsupportedOperationException: Unable to open DISPLAY
        at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.lambda$new$6(GtkApplication.java:181)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
        at javafx.graphics/com.sun.glass.ui.gtk.GtkApplication.<init>(GtkApplication.java:179)
        at javafx.graphics/com.sun.glass.ui.gtk.GtkPlatformFactory.createApplication(GtkPlatformFactory.java:41)
        at javafx.graphics/com.sun.glass.ui.Application.run(Application.java:146)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.startup(QuantumToolkit.java:291)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:293)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:163)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:659)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:679)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication$2(LauncherImpl.java:196)
        at java.base/java.lang.Thread.run(Thread.java:840)

> Task :interop-lab:demo-apps:run FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':interop-lab:demo-apps:run'.
> Process 'command '/usr/lib/jvm/java-17-openjdk-arm64/bin/java'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 8s
50 actionable tasks: 2 executed, 48 up-to-date

jithin-c-s, 2024-09-13:
Thanks a lot
I'm currently experiencing issues with my MP70 monitor(some hardware issue,. for the time being I'll check the CSV generation on 
a different device, the Fluke Prosim 8. I need to determine where the CSV files are being saved.

simonkelly1973, 2024-09-13:
Hi.  The "can't open display" message is really a Unix thing, rather than being Java specific. Essentially, a display in this context means a graphical display.  You have suggested that your objective is to run everything "headless".  Headless means "there is no graphics display".  So if you've got a machine where you are just connecting via ssh, then there is no "graphical display" associated with that login session.

If you have a different machine (like Mac or Linux), and you have appropriate permissions, you can configure the graphical display for your ssh session to be on a different machine.  The exact mechanism for that varies according to your setup, but commonly you either do

export DISPLAY=192.168.1.4:0.0

to configure graphics to be displayed on an X server that is running on 192.168.1.4, or you can configure an X tunnel using SSH.

To run the device interface standalone though, you don't need X.  You can just start the particular device interface from the command line.

simonkelly1973, 2024-09-16:
You might want to read this ticket, as it's closely related to your requirement, and we made some fixes based on the information we received.

https://github.com/mdpnp/mdpnp/issues/112

jithin-c-s, 2024-09-17:
need some help 
as i told last time , MP70 is having some issue , so i continued my testing with Fluke ProSim 8 

steps followed 

1. executed ./buildAndDeploy.sh , it created a bundle named OpenICE-1.5.0-SNAPSHOT.zip in mdpnp/interop-lab/demo-apps/build/distributions/
2. copied zip file parallel to mdpnp directory, and unzipped
3. OpenICE-1.5.0-SNAPSHOT/bin/OpenICE --help
[image]

5. then connected fluke prosim 8 to pi usb port 
6. then executed dmesg | grep tty,
**hmicro@hmicro-pie:~ $ dmesg | grep tty
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=DC:A6:32:D3:B9:9B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=ebe68f4f-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=IN
[    0.000349] printk: console [tty1] enabled
[    1.612649] fe201000.serial: ttyAMA1 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.612908] serial serial0: tty port ttyAMA1 registered
[    4.093756] systemd[1]: Created slice system-getty.slice - Slice /system/getty.
[    7.096124] usb 1-1.1: FTDI USB Serial Device converter now attached to ttyUSB0
[ 2262.732248] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[ 2268.675170] usb 1-1.1: FTDI USB Serial Device converter now attached to ttyUSB0
[ 2273.076741] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[ 2276.035375] usb 1-1.1: FTDI USB Serial Device converter now attached to ttyUSB0**

8. ttyUSB0 is the port number where fluke prosim is connected 
9. echo -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0 > device.this (as per the documentation in https://www.openice.info/docs/5_device-adapter-config.html)
10. rebooted pi
11.then executed command : systemctl status device-adapter
     hmicro@hmicro-pie:~ $ systemctl status device-adapter
     Unit device-adapter.service could not be found.

this means the Device-Adapter is not running ryt ? 
did i miss something ? 

if this device-adapter starts running after reboot , it should store the data from fluke ProSim 8 at some path ryt   ? how can i identify that path !

simonkelly1973, 2024-09-17:
There are several problems with that documentation - but the short version is it's very out of date, and very specific to the BeagleBone image that you found is now missing. - because we can't really support modern BeageBone machines anyway.

So - the longer version of that is that in that original BeagleBone image, a service was configured, and that service was called "device-adapter".  In other words, that service was part of the operating system on that image.  The device-adapter service read it's configuration from the file called "device.this" that you configured according to those instructions.

However, on your RaspberryPi, there is no custom image, and so there is not magically a service called "device-adapter" that is going to read that configuration file.

To run a device interface, you just need to do something like this

OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0

And that will start the device interface for the Fluke Prosim, using ttyUSB0.

However, this will only publish device data.  It does not write the device data anywhere.  To read the data yourself, you must subscribe to what is published.  That's what the Python code does.

Anyway, hopefully this at least gets the device to run in your environment, and then we can move on to the Python.

jithin-c-s, 2024-09-17:
hmicro@hmicro-pie:~/repo $ OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0
The library libnddsjava.so could not be loaded by Linux.
Make sure that the library is in your LD_LIBRARY_PATH environment variable.
Exception in thread "main" org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'iceQoS' defined in class path resource [RtConfig.xml]: Failed to instantiate [org.mdpnp.rtiapi.qos.IceQos$LoadStatus]: Factory method 'loadAndSetIceQos' threw exception with message: no nddsjava in java.library.path: /usr/java/packages/lib:/usr/lib/aarch64-linux-gnu/jni:/lib/aarch64-linux-gnu:/usr/lib/aarch64-linux-gnu:/usr/lib/jni:/lib:/usr/lib
	at org.springframework.beans.factory.support.ConstructorResolver.instantiate(ConstructorResolver.java:652)
	at org.springframework.beans.factory.support.ConstructorResolver.instantiateUsingFactoryMethod(ConstructorResolver.java:488)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.instantiateUsingFactoryMethod(AbstractAutowireCapableBeanFactory.java:1324)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBeanInstance(AbstractAutowireCapableBeanFactory.java:1161)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:561)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:521)
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:326)
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:324)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:200)
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:961)
	at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:915)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:584)
	at org.mdpnp.apps.testapp.Configuration.createContext(Configuration.java:179)
	at org.mdpnp.apps.testapp.DeviceAdapterCommand.execute(DeviceAdapterCommand.java:74)
	at org.mdpnp.apps.testapp.Main.main(Main.java:52)
Caused by: org.springframework.beans.BeanInstantiationException: Failed to instantiate [org.mdpnp.rtiapi.qos.IceQos$LoadStatus]: Factory method 'loadAndSetIceQos' threw exception with message: no nddsjava in java.library.path: /usr/java/packages/lib:/usr/lib/aarch64-linux-gnu/jni:/lib/aarch64-linux-gnu:/usr/lib/aarch64-linux-gnu:/usr/lib/jni:/lib:/usr/lib
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:171)
	at org.springframework.beans.factory.support.ConstructorResolver.instantiate(ConstructorResolver.java:648)
	... 15 more
Caused by: java.lang.UnsatisfiedLinkError: no nddsjava in java.library.path: /usr/java/packages/lib:/usr/lib/aarch64-linux-gnu/jni:/lib/aarch64-linux-gnu:/usr/lib/aarch64-linux-gnu:/usr/lib/jni:/lib:/usr/lib
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2434)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:818)
	at java.base/java.lang.System.loadLibrary(System.java:1993)
	at com.rti.dds.util.NativeInterface.loadNativeLibrary(NativeInterface.java:153)
	at com.rti.dds.util.NativeInterface.loadNativeLibraries(NativeInterface.java:125)
	at com.rti.dds.domain.DomainParticipantFactory.<clinit>(DomainParticipantFactory.java:78)
	at org.mdpnp.rtiapi.qos.IceQos.loadAndSetIceQos(IceQos.java:47)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:569)
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:139)
	... 16 more

simonkelly1973, 2024-09-17:
This happens because of the shared library it mentions not being on your path.  When things are run through gradle, all this is configured automatically because it's in the gradle scripts.  In the meantime, you can do

export LD_LIBRARY_PATH=....

but we can't tell from this output where the ... should be.

You should be able to see, underneath the clone of the repository, the directory

interop-lab/demo-apps/native-libs/aarch

Whatever the full path to that directory is, you should set as the variable value.  For instance

export LD_LIBRARY_PATH=/home/simonkelly1973/mdpnp/interop-lab/demo-apps/native-libs/aarch

Then re-run the original command.

jithin-c-s, 2024-09-17:
hmicro@hmicro-pie:~/repo $ OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0
Exception in thread "main" org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'iceQoS' defined in class path resource [RtConfig.xml]: Failed to instantiate [org.mdpnp.rtiapi.qos.IceQos$LoadStatus]: Factory method 'loadAndSetIceQos' threw exception with message: 'int com.rti.dds.infrastructure.MonitoringLoggingForwardingSettings.get_native_security_event_forwarding_levelI(long)'
	at org.springframework.beans.factory.support.ConstructorResolver.instantiate(ConstructorResolver.java:652)
	at org.springframework.beans.factory.support.ConstructorResolver.instantiateUsingFactoryMethod(ConstructorResolver.java:488)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.instantiateUsingFactoryMethod(AbstractAutowireCapableBeanFactory.java:1324)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBeanInstance(AbstractAutowireCapableBeanFactory.java:1161)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:561)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:521)
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:326)
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:324)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:200)
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:961)
	at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:915)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:584)
	at org.mdpnp.apps.testapp.Configuration.createContext(Configuration.java:179)
	at org.mdpnp.apps.testapp.DeviceAdapterCommand.execute(DeviceAdapterCommand.java:74)
	at org.mdpnp.apps.testapp.Main.main(Main.java:52)
Caused by: org.springframework.beans.BeanInstantiationException: Failed to instantiate [org.mdpnp.rtiapi.qos.IceQos$LoadStatus]: Factory method 'loadAndSetIceQos' threw exception with message: 'int com.rti.dds.infrastructure.MonitoringLoggingForwardingSettings.get_native_security_event_forwarding_levelI(long)'
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:171)
	at org.springframework.beans.factory.support.ConstructorResolver.instantiate(ConstructorResolver.java:648)
	... 15 more
Caused by: java.lang.UnsatisfiedLinkError: 'int com.rti.dds.infrastructure.MonitoringLoggingForwardingSettings.get_native_security_event_forwarding_levelI(long)'
	at com.rti.dds.infrastructure.MonitoringLoggingForwardingSettings.get_native_security_event_forwarding_levelI(Native Method)
	at com.rti.dds.infrastructure.MonitoringLoggingForwardingSettings.pull_from_nativeI(MonitoringLoggingForwardingSettings.java:128)
	at com.rti.dds.infrastructure.MonitoringTelemetryData.pull_from_nativeI(MonitoringTelemetryData.java:111)
	at com.rti.dds.infrastructure.MonitoringQosPolicy.pull_from_nativeI(MonitoringQosPolicy.java:72)
	at com.rti.dds.domain.DomainParticipantFactoryQos.pull_from_nativeI(DomainParticipantFactoryQos.java:270)
	at com.rti.dds.infrastructure.NativeStructMixin.get_untypedI(NativeStructMixin.java:115)
	at com.rti.dds.domain.DomainParticipantFactoryImpl.get_qos(DomainParticipantFactoryImpl.java:561)
	at org.mdpnp.rtiapi.qos.IceQos.loadAndSetIceQos(IceQos.java:49)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:569)
	at org.springframework.beans.factory.support.SimpleInstantiationStrategy.instantiate(SimpleInstantiationStrategy.java:139)
	... 16 more

jithin-c-s, 2024-09-17:
hmicro@hmicro-pie:~/repo/mdpnp $ sh ./buildAndDeploy.sh 
NOTE: To build with OpenSplice compatibility please set the OSPL_HOME and SPLICE_TARGET environment variables.

> Configure project :data-types:x73-idl-rti-dds
rtiddsgenExplodeResources started
rtiddsgenExplodeResources ended

> Configure project :interop-lab:demo-apps
**Project :interop-lab:demo-apps => no module-info.java found**

BUILD SUCCESSFUL in 57s
52 actionable tasks: 4 executed, 48 up-to-date
Shipping OpenICE-1.5.0-SNAPSHOT bundle to 10.0.1.200

is **Project :interop-lab:demo-apps => no module-info.java found** can cause any issue ?

simonkelly1973, 2024-09-17:
Hello

The stack trace looks like an issue with a difference between a native library and Java code.  We did make changes to this in the last few days.

Could you do a

git pull

Just to check you are up to date please?  It might show that the .so files are replaced.  Then try again.

simonkelly1973, 2024-09-17:
Yes - if you use a tag, later changes are not in the tagged version.

You should just be able to do

git pull
git checkout master

jithin-c-s, 2024-09-17:
i checked out to master ,

then im getting the following error 

hmicro@hmicro-pie:~/repo $ OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0
RTI Connext DDS No source for License information
Please contact support@rti.com with any questions or comments.
Exception in thread "main" org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'subscriber' defined in class path resource [RtConfig.xml]: Cannot resolve reference to bean 'domainParticipant' while setting constructor argument
	at org.springframework.beans.factory.support.BeanDefinitionValueResolver.resolveReference(BeanDefinitionValueResolver.java:377)
	at org.springframework.beans.factory.support.BeanDefinitionValueResolver.resolveValueIfNecessary(BeanDefinitionValueResolver.java:135)
	at org.springframework.beans.factory.support.ConstructorResolver.resolveConstructorArguments(ConstructorResolver.java:700)
	at org.springframework.beans.factory.support.ConstructorResolver.autowireConstructor(ConstructorResolver.java:211)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.autowireConstructor(AbstractAutowireCapableBeanFactory.java:1344)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBeanInstance(AbstractAutowireCapableBeanFactory.java:1188)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:561)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:521)
	at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:326)
	at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:324)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:200)
	at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:955)
	at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:915)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:584)
	at org.mdpnp.apps.testapp.Configuration.createContext(Configuration.java:179)
	at org.mdpnp.apps.testapp.DeviceAdapterCommand.execute(DeviceAdapterCommand.java:74)
	at org.mdpnp.apps.testapp.Main.main(Main.java:52)
Caused by: org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'domainParticipant': FactoryBean threw exception on object creation
	at org.springframework.beans.factory.support.FactoryBeanRegistrySupport.doGetObjectFromFactoryBean(FactoryBeanRegistrySupport.java:154)
	at org.springframework.beans.factory.support.FactoryBeanRegistrySupport.getObjectFromFactoryBean(FactoryBeanRegistrySupport.java:90)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getObjectForBeanInstance(AbstractBeanFactory.java:1823)
	at 
org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.getObjectForBeanInstance(AbstractAutowireCapableBeanFactory.java:1265)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:259)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:200)
	at org.springframework.beans.factory.support.BeanDefinitionValueResolver.resolveReference(BeanDefinitionValueResolver.java:365)
	... 17 more
Caused by: com.rti.dds.infrastructure.RETCODE_ERROR: error creating entity
	at com.rti.dds.util.Utilities.rethrow(Utilities.java:93)
	at com.rti.dds.infrastructure.NativeFactoryMixin.create_entityI(NativeFactoryMixin.java:187)
	at com.rti.dds.domain.DomainParticipantFactoryImpl.create_participant(DomainParticipantFactoryImpl.java:287)
	at org.mdpnp.devices.DomainParticipantFactory.getObject(DomainParticipantFactory.java:111)
	at org.mdpnp.devices.DomainParticipantFactory.getObject(DomainParticipantFactory.java:25)
	at org.springframework.beans.factory.support.FactoryBeanRegistrySupport.doGetObjectFromFactoryBean(FactoryBeanRegistrySupport.java:148)
	... 23 more

jithin-c-s, 2024-09-17:
look like the license file is present in **mdpnp/interop-lab/demo-apps/src/main/resources** path

simonkelly1973, 2024-09-17:
Now you need to configure license information.  You need to do

export RTI_LICENSE_FILE=...

Just as with the  LD_LIBRARY_PATH, we can't know exactly what the path is.  However, the required license file is in the source distribution you cloned.  The file you need is

interop-lab/demo-apps/src/main/resources/OpenICE_license.dat

so just do

export RTI_LICENSE_FILE=/home/simonkelly1973/mdpnp/interop-lab/demo-apps/src/main/resources/OpenICE_license.dat

(to use the same example as before).  Anyway, that's the variable you need to export, with a full path to that file.

jithin-c-s, 2024-09-17:
look like it is working, im not getting any error 

hmicro@hmicro-pie:~/repo $ OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 15 -device FlukeProsim68 -address ttyUSB0
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only with OpenICE. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.

simonkelly1973, 2024-09-17:
OK that looks great.

So, however you might want to implement it, if you want to autostart this when your machine boots up, you can add something to your init scripts or whatever that just runs that one command.

simonkelly1973, 2024-09-17:
Now you can consider getting the Python working.  You might have noticed via the other ticket I highlighted yesterday that we've now updated the Python example directory with a README, but the quickstart based on your existing device interface should be

pip install rti.connext
export RTI_LICENSE_FILE=.....       (use exactly the same value as before)
python read_numerics.py -d 15   (use domain 15, as you have done for the device startup)

jithin-c-s, 2024-09-17:
Thank you for the update! I appreciate the guidance.I will proceed with setting up the Python environment as suggested.

jithin-c-s, 2024-09-18:
Hi, 

look like the python script is running, but its not reporting the values 

[image]

simonkelly1973, 2024-09-18:
Hello

What are you attempting to do with the Fluke?  That device is designed to drive data to another device such as a patient simulator.

Let's try running your Python code in the same way but with a different device.  Can you run the device interface code but with a different device like the simulated pump?  That produces a single numeric once per second.

jithin-c-s, 2024-09-18:
I hadn’t planned to use the Fluke device, but since the MP70 isn’t functioning at the moment, I decided to begin with a device that's listed as supported.
I have a Masimo Rad-97 available. Can I use it?

simonkelly1973, 2024-09-18:
We should use a simulated device to test the Python code first, because that rules out any problem with the connection to a physical device.

Once we know that that is working we can concentrate on getting a physical device working.

jithin-c-s, 2024-09-18:
with device as Controllable_Pump

[image]

jithin-c-s, 2024-09-18:
few doubts 

1. i could see that FlukeProsim68 is supported under device , what is the use of that ? control Fluke and change input values or read the values from the Fluke ? (i understand it is a signal generator which will be fed to a monitor, and we are supposed to retrieve the data  from monitor)
2. in Philips Intellivue series only MP70/MP60 seems to be supported , is there any chance it will work with IntelliVue MX750 (https://www.philips.co.in/healthcare/product/HC866471/intellivue-mx750-bedside-patient-monitor - planning to get a demo device if it supports)
3. **Or can you suggest a monitor that fully supports Invasive BP, ECG, and SpO2 together **? In that case, it would be easier to configure and use .
4.while checking OpenICE launch Settings , under device interface Nihon Koden is present , is it supported by device adapter ?  
[image]

simonkelly1973, 2024-09-18:
OK, so the Python side is working correctly.  Let's look now at the devices you mentioned.

MP70 - this is out of action at the moment.

FlukeProsim8 - the reason you do not see any numeric values for this is that Fluke ProSim device interface does not produce any numeric values.  As I'm sure you are aware, the purpose the Fluke ProSim itself is to provide controllable inputs to another device - such as the MP70.  The MP70 (or equivalent Intellivue device) would publish numeric values for the metrics it is measuring.  But the ProSim itself does not.  The purpose of the ProSim device interface is to allow OpenICE to control the ProSim, which in turn provides the controlled data values to the MP70, which in turn can publish numerics that OpenICE can receive.

The short version of all that is that the ProSim does not produce numerics, so you don't see it do anything in the Python code.

Masimo Radical 97 - The supported version name is "Radical 7" - but the protocol is very simple so if the "Radical 97" uses the same protocol, you should certainly see values from that.  It supports pulse and Sp02.

jithin-c-s, 2024-09-18:
`Masimo Radical 97 - The supported version name is "Radical 7" - but the protocol is very simple so if the "Radical 97" uses the same protocol, you should certainly see values from that. It supports pulse and Sp02.`

for radical 97,  pi is not detecting USB serial port , tried connecting to RJ45 port as well as USB port to pie
in both case it is not being detected by pi
only option left is to directly connect the tx and rx coming from RJ45 to serial pins in pi (not sure whether it will work or not )

simonkelly1973, 2024-09-18:
MX750 does support Intellivue, yes, so should work with OpenICE.

The NKV550 is supported by OpenICE, but it requires a level of firmware on the device that is not publically available.  This driver was a research collaboration with Nihon Koden.

jithin-c-s, 2024-09-18:
please find the attached 

[demo-apps.log](https://github.com/user-attachments/files/17044140/demo-apps.log)

jithin-c-s, 2024-09-18:
so the only options will be MP70 and  MX750

simonkelly1973, 2024-09-18:
I don't know if MX750 has been tried - in our lab we use MX800, but I think you should be very confident that the 750 would work because they all use the Intellivue protocol.

In your demo-apps.log, there is no reference to the Radical class.  It does get as far as loading the serial provider class, but we can't see anything more than that at the moment.

jithin-c-s, 2024-09-20:
still  waiting for the monitor to arrive.
could you please provide an example of how to control the Fluke ProSim 8?

simonkelly1973, 2024-09-20:
Hi.

So, the idea of the ProSim 8 is that the device interface in OpenICE listens for what in OpenICE, we call "Global Simulation Objectives".  This means that you can create the ProSim device, and then run the "Simulation Control" application in OpenICE

[image]

You can then adjust these different values like SpO2, pulse and so on, and the Fluke device interface in OpenICE should detect those changes, and then control your Fluke ProSim to have the same values for those properties that you have just set in the application.

jithin-c-s, 2024-09-30:
just got MP70 back, 
can i follow the bellow diagram for the connection ? 
[image]
MP70 have RJ45 port - can i use LAN cable for the connection between raspberry pi and MP70
and use below command to start 
"OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 0 -device IntellivueEthernet"

simonkelly1973, 2024-09-30:
Yes, you should be able to do that.  Let us know how it goes.

jithin-c-s, 2024-09-30:
DB 9 connector was not available 
so directly connected RJ45  to raspberry pi Pins
[image]

RJ45 PIN 4(GND) to  RPI (PIN 6 -GND)
RJ45 PIN 5(RXD) to RPI (PIN 8 -GPIO 14 (TXD))
RJ45 PIN 7(RXD) to RPI (PIN 10 -GPIO 15 (RXD))

with this connection i started device interface on one terminal - 
"/home/$USER/jithin/repo/OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 0 -device IntellivueEthernet"

and read_numerics on another terminal 
"/home/$USER/jithin/env/python_3/bin/python /home/$USER/jithin/repo/mdpnp/data-types/x73-idl/src/main/idl/ice/numeric_subscriber/read_numerics.py -d 0"

but im not receiving any data
attaching demo-apps logs

[demo-apps.log](https://github.com/user-attachments/files/17189603/demo-apps.log)

jithin-c-s, 2024-09-30:
ok

i tried enabling uart in RPI, but for some reason im  not able to find the address
so i will get a serial adapter 
Could you please suggest a DigiKey Part Number for a serial connector that is compatible with both the Raspberry Pi and Philips monitors?

jithin-c-s, 2024-10-01:
got the serial cables ,
setup -
RJ45 to DB9(Female) - DB9( Male ) to USB -> RPI USB PORT

i used below commands and tried to get the data 

OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 0 -device IntellivueEthernet -address ttyUSB0
OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 0 -device IntellivueSerial -address ttyUSB0

in both case im not receiving the data
any idea why  ?

jithin-c-s, 2024-10-01:
otherwise can you share some diagram for the connection  ?

jithin-c-s, 2024-10-01:
while going through the app logs , saw some ERRORS

attaching the logs 
[demo-apps.log-7.log](https://github.com/user-attachments/files/17201125/demo-apps.log-7.log)

jithin-c-s, 2024-10-01:
[image]

look like the RJ45 is not labeled as either RS232, MIB or an arrow pointing out from a circle
but the HD-15 connector is marked with an arrow pointing out from circle

simonkelly1973, 2024-10-01:
The HD-15 is also labelled "Video", which would be the normal use of such a connector.  Here is a link to the service manual for an MP60/MP70, that might help you to identify if you have a suitable connection available.

[Service Manual](http://www.frankshospitalworkshop.com/equipment/documents/ecg/service_manuals/Philips_IntelliVue_MP60_-_Service_Manual.pdf)

jithin-c-s, 2024-10-01:
i didn't understand much from it , but look like its is using RS-422 standard for communication (not RS-232, and im using RS232 cable )
but not sure the data can be exported from RJ45/Video port

jithin-c-s, 2024-10-01:
I’ll be receiving the MX-450 in two days, and I’m hoping the connection process will be smooth and straightforward.
maybe you can share the connection for MX Series here (if its available )

meantime i will explore MP70 
are you still using MP70 in your lab ? what is the connection you are using for data collection

jithin-c-s, 2024-10-03:
HI,

i was trying different way for getting the data
followed the below steps 

1. used laptop instead of RPI
2. connected lan cable from MX-500 to Router
3. laptop and router connected to same network
4. ./gradlew run

 

created ICE device adapter 
Device Category - network
device type philips intellivue (LAN)
provided the ip address of MX-500

[image]

[image]

with the above settings  im able to see the waveform in the gui

why this is not working in RPI ? any idea ? 
did i miss any steps  ?

simonkelly1973, 2024-10-03:
Using the lan port connected to a hub and using the IP address of the Philips device is very different to trying to use serial communications.  Are you saying that the same LAN/IP setup does not work on the Raspberry Pi?

jithin-c-s, 2024-10-03:
in rpi i couldn't  find any argument to parse ip address , when IntellivueEthernet selected as device.

simonkelly1973, 2024-10-03:
`OpenICE-1.5.0-SNAPSHOT/bin/OpenICE -app ICE_Device_Interface -domain 0 -device IntellivueEthernet -address 192.168.0.103`

So when using a network device, just use the -address argument to specify the IP address.

[40 more replies]