mdpnp/mdpnp #114: Scenario Control

rhepstein1, 2024-09-17:
Are there instructions for using the Scenario Control app in the ICE Applications Supervisor dialog?
Also for some of the other apps:
Data Visualization

Also, do you have an estimate as to when the Data Exporter app will be displayed and what will need to be done to make it appear?

simonkelly1973, 2024-09-18:
Hello

If you do a

git pull

against the master branch in GitHub, you should have a working Data Exporter app.  We still need to investigate the root cause of this problem, but for now the previously uncaught exception is caught and logged where it occurs, and this allows other applications to start normally.

We were able to create a CSV file in our tests.

rhepstein1, 2024-09-19:
I downloaded the current files in the Master branch and recompiled and the Data Recorder is now visible and exports to csv.

Using the simulated multiparameter, the waveform morphologies are correct, but the numbers don't correspond to the physical units (e.g., mV for ECG and mm Hg for blood pressure). Can you explain what those values correspond to?

I couldn't find any documentation for the csv output columns, which were easy to figure out, but it would be helpful to either add the column names as the first row in the output or to provide documentation.

[image]

simonkelly1973, 2024-09-19:
Well, we are dangerously close to me straying out of the software domain and into bio-med or clinical issues here, which Julian and others will confirm usually has fairly hilarious results.  However, I think for these purposes, the short answer is that the values themselves do not correspond to anything, with the focus really just being on relative values between different values in the waveform.

If you're able to look at the code, the relevant file for the sources of the ECG waveform (for example) are

interop-lab/demo-devices/src/main/java/org/mdpnp/devices/simulation/multi/SimMultiparameter.java
and
devices/simulated/src/main/java/org/mdpnp/devices/simulation/ecg/SimulatedElectroCardioGram.java

On line 102 in the first of those files, it has the comment

`// TODO get better numbers in actual millivolts`

The base inputs for the initial values are hard coded into the second file, and are just a long array of floating point values like this

1754.6228740250176, -1.8702723856853978, -17.30350351479403, 0.9639533281850988,
            3.840507863935154, 7.934373158467186, -6.404393385136917, -11.414941750464283...

Those are then manipulated using a routine in the file

/common/src/main/java/org/mdpnp/devices/math/DCT.java

to produce the final values that are actually plotted.  However, as you can see from the very first comment, the intention was to get "better" values.

There is a contrast with at least some of the real physical devices that I'm familiar with.  When using a Philips MX-800 (that we have in the lab) and obtaining arterial blood pressure, if the BP is 120/80, then the numeric value of the highest and lowest points in the waveform data are 120 and 80 respectively.

rhepstein1, 2024-09-19:
> CAUTION: This email originated from outside the organization. DO NOT CLICK ON LINKS or OPEN ATTACHMENTS unless you know and trust the sender.

That’s just fine if when connecting to an actual monitor the numbers are correct, as you describe. Since the waveforms on the screen don’t change in response to altering the numeric in the simulation control app, what you are describing as a canned set of values makes sense.

The main issue is that the data recorder is producing output and is not appearing in the app list.

Rich

Richard H. Epstein, MD
Professor of Clinical Anesthesiology
Miller School of Medicine
University of Miami

Please excuse my brevity/typos.

rhepstein1, 2024-09-19:
should have been “.. and is appearing …”
That is, it’s working.