mdpnp/mdpnp #116: Creating OpenICE Device Adaptor on Raspberry Pi 5

rhepstein1, 2024-09-18:
Do you have detailed instructions for installing OpenICE v 2.0 on RPi  for use as a device adapter?

Also, are there instructions on connecting the RPi via a home router to a windows computer on which OpenICE supervisor is running.

There are bits and pieces scattered in various places but it’s hard to figure out exactly what to do.

rhepstein1, 2024-09-19:
Any more info about this? My RPi and pocket router just arrived and I want to get these set up.

simonkelly1973, 2024-09-19:
It's pretty straightforward to get an OpenICE version running on RaspberryPi.  The good news is that no particularly special effort is required to compile and run the full product on a Pi.  You can just do a git clone and it will compile and run - BUT there is one change required.

Edit the following two files

./interop-lab/demo-guis-javafx/build.gradle
./interop-lab/demo-apps/build.gradle

Both of those contain a line

version = "17.0.11"

Just edit that to

version = "17.0.8"

(that's all because the required Java libraries don't always seem to get compiled for the correct architecture for a Raspberry Pi)

That takes care of being able to compile and run on the Pi.

In terms of being able to access the resulting data by running a separate instance of OpenICE on Windows, that should work fine assuming everything is on the same subnet.  The biggest possible issue is that a firewall on the windows side blocks incoming traffic from the Pi.  That's configurable though - we can check the relevant ports on the Pi side that are being used for transmitting the data and open the same ones on Windows.

Do you want to connect a physical device to the Pi using a serial connector?  If so the correct way to go is to buy a USB<->Serial convertor, that means you can plug a a USB cable into the Pi and the other end of the cable is a 9 pin serial adapter.  We use that successfully for many devices in the lab.  There are a lot of variants of this type of cable but the most reliable we have found are "official" FTDI cables (FTDI are the chip manufacturer).

Going back a long way in the history of the project, we used to have a full "SD card image" for the BeagleBone so that people could just install the entire operating system and the application, libraries etc. onto a SD card and have a working system.  The main reason for that though was that it was very complicated to piece together a working installation of Java and the dependencies for a BeagleBone board.

Since the app can be compiled and run on a Raspberry Pi without any special issues (apart from that small revision to the JavaFX version) we haven't produced an equivalent SD Card image for the Pi.

Do let us know how you are getting on and if there are any other questions.

rhepstein1, 2024-09-20:
I am having trouble getting the right versions of javafx installed on my Raspberry Pi. The device came without java.

CPU: Broadcom  BCM2712 quad core ARM Cortex A76 processor @2.4 GHz
OS: Raspbian GNU/linux 12 (bookworm)  This was as delivered, the 32-bit versin
CPU:   aarch64   (64-bit ARM processor)
 

I installed openjde version 17  via   git apt install openjde17-jde -y  
and added the /usr/bin/java location to my path
java –version confirms that version 17 of the jde is installed.
So this looks OK.

However, I can’t find a version of OpenJFX that works on my device. 
I’ve downloaded several which report as installing properly, but when I try to compile openice I get various error messages related to not being compatible with my hardware.

Can you supply a link to where I can get a version of javafx 17.0.8 that will work with openice or other advice on how to get my RPi working?

 I have a working serial to usb cable I use to download from a  Philps MX550 monitor.
The only thing that I’m missing is getting the device connector built on the RPi

simonkelly1973, 2024-09-20:
It's a curious quirk of the most modern Raspberry Pi systems that they have 64 bit hardward but ship with a 32 bit OS by default.  Java is not well supported on 32 bit systems nowadays.  The first step is to switch to the 64 bit version of the operating system.

The reason you are having the problem with OpenJFX is the 32/64 bit architecture problem.

When you compile OpenICE, it will use the Gradle build system to automatically download and use all the relevant dependencies - including OpenJFX.  So you don't need to download that yourself.  The reason you are seeing compile problems is the same reason that the build will fail as well - there is no compatible build of OpenJFX for "32 bit aarch" architecture.

We can be sure it definitely all works with the 64 bit version of the OS.

simonkelly1973, 2024-09-20:
(Sorry - hadn't intended to close this!)

rhepstein1, 2024-09-20:
Thank you. I'll replace the OS and try again.

rhepstein1, 2024-09-21:
After converting the RPi OS to 64 bit Debian, was able to compile the software and get OpenICE running.

closed 2024-09-21 by rhepstein1