mdpnp/mdpnp #118: Running OpenICE Device and Python on a Raspberry Pi

rhepstein1, 2024-09-21:
Is it possible to connect the OpenICE Device Adapter output to python code running on the same RPi device, or does this require a network connection to a separate computer running the OpenIcE supervisor?

My use case only involves a single patient monitor with no need to integrate multiple devices.

simonkelly1973, 2024-09-21:
Yes, you can run the Python code and the device interface on the same device.

The device interface publishes data.  The Python code subscribes to it.  Those two processes can be on the same machine or a different one.

rhepstein1, 2024-09-21:
I have python 3.11 on my RPi as the installed version. I ran simple scripts not involving any openice components and those execute fine. So I think my python install is fine.

However, when I try to install rti.connext, I get an error that the environment is being externally managed and it won't load.
In the python 3.11 folder, I tried:
python -m pip install rti.connext 

That worked on my windows machine, but not on the RPi.

I think I need rti.connext because the system complained when I tried to run the numeric_subscriber.py script in the sample python folder.

rhepstein1, 2024-09-22:
I was able to create a virtual environment that I called my-env to download  rti.connext.
The python 3.11 binaries and other files are there, and rti is under that.

How do I get the openice script to execute using that version of python.exe vs the global one in the path?

I still have to set the license file, but that is just adding it to /etc/environment and rebooting.

rhepstein1, 2024-09-22:
There is probably something simple I'm missing. I know just enough linux to be dangerous.
What is the directory I need to be in and what is the syntax to launch one of the Numeric.py sample program?

I activated my virtual environment with source my-venv/bin/activate which brings up the prompt showing I'm in my-env
Under that are copies of python, rti, etc.

I tried running 
python /path to the folder with Numeric.py/Numeric.py

I then get the error that it can't find the module rticonnextdds_connector 
That the first line in the Numeric.py script so that is opening.

Do I need to add something to the $PATH variable?

I'm not getting an error about the license, so I think that what I set for $RTI_LICENSE_FILE is likely ok.

rhepstein1, 2024-09-22:
I figured it out.
The sample python code in /interop-labs/python doesn't work (at least on my machine).
The working code is in a different folder

I copied
ice.py
numeric_subscriber.py

from
/data-types/x73-idl/src/main/idl/ice/numeric_subscriber

into the folder where I'm running my python scripts
 
from the root of my virtual environment, I just ran
python /{python script folder}/numeric_subscriber.py

and it connects to the OpenIce Device connector and displays the values

Woo Hoo!

The next step is plugging in my RJ45 to Serial to USB cable into the back of my Philips MX550 monitor and the RPi and see if I can collect live data rather than from the simulation.

simonkelly1973, 2024-09-22:
That seems to be more than one issue.

The code in interop-labs/python is older code that we wrote for some different purposes before RTI had their own full feature integration with Python.

Successfull completion of

pip install rti.connext

is required to use the numeric subscriber from

data-types/x73-idl/src/main/idl/numeric_subscriber

which is the most up-to-date Python integration that uses RTI's own Python integration.  Without seeing the output from your attempt to install rti.connext the first time, it's hard to say.  That might well be a permissions problem, with the global Python module installation directory not being writable for non-root user.  That permissions constraint doesn't apply to a virtual environment.  The "virtual environment" concept is a Python thing, not a Linux thing.

Anyway, we will look forward to seeing your results with the MX550.

rhepstein1, 2024-09-22:
it was a python warning about possibly breaking the python install by installing the module. Similar warnings were produced when I tried to install other external modules. I have root permissions. Instructions were provided on how to override and install the modules anyway, but it seemed less risky to just create a virtual environment. I think since python 3.11 is part of the Debian image, there is a theoretical risk of an issue if the OS is updated. 

You might consider deleting code from the latest version that doesn’t work. It’s confusing, especially when the working code is parked in a somewhat obscure directory and there is a high level directory named python with .py examples.

closed 2024-09-22 by rhepstein1