mdpnp/mdpnp #119: Device Support

rhepstein1, 2024-09-23:
Do you currently support or have plans to support any of the following devices:

1. GE Aisys Anesthesia Machines
2. Draeger Anesthesia Machines more recent than Apollo
3. Quantitative Neuromuscular Block Monitors (TwitchView, TetarGraph, or others)
4. Mindray Anesthesia Machines
5. Edwards Hemisphere monitor

simonkelly1973, 2024-09-23:
I'm happy to ask in the project what our next device support plans are.  However, looking at the other devices.

3.  Tetragraph appears to have a bridge to IntelliVue, so that's a possible route.
4.  Mindray A Series seems to have a well documented interface (A Series Comms Protocol), so it would also appear to be possible to make a device interface for that.
5.  Edwards Hemosphere appears to have several communication interface options, including HL7, so that's a possible interface option.

Just to be clear - I am not in a position to commit to doing any of this work, although I'm sure Julian would be happy to hear any proposals you'd like to make about providing support for these devices.

rhepstein1, 2024-09-23:
For protocols you currently support, are the parameters received only those hard coded in the interface, or do you pick up anything sent across? For example, if an Intellivue Module is added to a Philips monitor and the information from the connected monitor shows up on the monitor screen, would the parameter be received by the interface and recorded by the Data Recorder app. Or do you have to add in the parameter name and perhaps map it to the field being transmitted, such as MDC_PRESS_BLD_ART_ABP for the blood pressure waveform or MDC_PULS_OXIM_SAT_O2 for the pulse oximeter oxygen saturation?

I will talk with Julian. I'm sure he'll want a use case to justify the development effort.

simonkelly1973, 2024-09-23:
It's not guaranteed that everything that the monitor can possibly produce is catered for in OpenICE.  There are several places additions may be required.  If you look at the file 

demo-devices/src/main/resources/org/mdpnp/devices/philips/intellivue/intellivue.map

That should give you an indicator of what would need to be added.  To determine the correct value for the first column, you ideally need access to the Intellivue specification - although with some knowledge of the OpenICE code and where to look, we can detect metrics that are being reported by the device that the OpenICE code doesn't recognise.

rhepstein1, 2024-09-23:
Thanks.

If I understand:
Column 1: Is this the Intellivue parameter name?
Column 2: Is this the OpenIce paramter name (as listed in the output from the OpenICE Data Recorder app)
Column 4: Parameter type (W=wave, N=numeric)

What is Column 3 how does it relate to Column 1 when the value in column 3 is not the value in column 1 prefixed by NLS_

If I were to know the Intellivue Name for a parameter not currently listed and add in the columns corresponding to the other rows, would that allow the data to be processed by OpenICE and who up in the Data Recorder output?

Is an example of values you are processing not through this list
MDC_PRESS_BLD_ART_ABP_SYS
MDC_PRESS_BLD_ART_ABP_DIA

I only see the noninvasive version.

simonkelly1973, 2024-09-23:
It's helpful that there is a copy of the Philips document available here

https://www.documents.philips.com/doclib/enc/fetch/2000/4504/577242/577243/577247/582636/582882/X2%2C_MP%2C_MX_&_FM_Series_Rel._L.0_Data_Export_Interface_Program._Guide_4535_645_88011_(ENG).pdf

If we take the line 14 in the map, it's this

NOM_PRESS_BLD_NONINV	MDC_PRESS_BLD_NONINV	NLS_NOM_PRESS_BLD_NONINV	N

Looking at that in the Philips manual, the definition is

[image]

So - Column 1 refers to the "Observed Value".  Column 3 refers to the "Label".  Column 2 is normally an MDC code - MDC meaning Medical Device Communications, which is part of 11073.  There are some codes (not necessarily from Intellivue, but from other devices) where there isn't an obvious relationship to an MDC code.  In that case, some "Ice specific" code like ICE_.....  is used instead.  The MDC code (or ICE code where an MDC code is not applicable) is used as the "metric_id" field in a published numeric or waveform.  

Column 4 just indicates whether the metric is a waveform or a numeric.

Regarding the arterial blood pressure - this is available via NOM_PRESS_BLD_ART_ABP .  The NOM_PRESS_BLD_ART_ABP_SYS, NOM_PRESS_BLD_ART_ABP_DIA and NOM_PRESS_BLD_ART_MEAN are compound values.

Exactly which metric the MX-800 produces depends on its operating mode.  We notice this from time to time in the MDPNP lab, after restarting the MX-800.  The ABP is not available because it's not being produced by the monitor in whatever operating mode its in.  Unfortunately I don't personally know how to change this operating mode myself as that's more of a bio-med thing.

rhepstein1, 2024-09-23:
Thanks.
Here is a link to the MDC codes that are not behind a paywall
https://termgit.elga.gv.at/CodeSystem-mdc-medicaldevicecommunications 

Typically, the code to get into the operating modes menu on a Philips monitor is 14432 unless someone changes it.
Select operating modes|Demo and the code
That will display the simulated waveforms and numbericcs on the monitor and send everything out the bus.
This should not be done when monitoring an actual patient.

If you add additional modules to the monitor, they will then show up on the screen and get output via the interface.
I haven't tested this with OpenIce, but it works like this when using the VitalRecorder program.
For example, if you plug in the sidestream capnograam module, the digitized capnograph and the associated values is produced.
If that module is not plugged in, then the values are not transmitted.

simonkelly1973, 2024-09-24:
Hello.  We do have other people who change the operating mode.  It's just that I'm not physically based in the lab, but work remotely, so it's not something I ever do myself.

We have recently made changes to the code for someone who is doing anaesthesia experiments, because they wanted to use an ECG module.  That worked OK after the changes, so it's definitely possible to amend OpenICE accordingly.