mdpnp/mdpnp #122: Simualted Controllable BP Monitor not Graphing Correctly

rhepstein1, 2024-09-27:
I'm not even sure what the purpose is for this device, but when graphing, you are missing code to erase the previous trace when the waveform wraps.

[image]

rhepstein1, 2024-09-27:
BTW, I changed the background in the CSS file as you suggested.

simonkelly1973, 2024-09-27:
We created this device during Covid, when we expected the lab to be closed for a few weeks (rather than the major long term disruption it became).  We were working on an application at the time that needed BP data, so we made a simulated device by capturing a recording from a real device.

The graphing is always independent of the device, so the problem is not related to "missing code to erase the previous trace".  The specific issue here is that the advertised frequency of the device data (i.e. the number of points in each chunk of the waveform) is not exactly the same as the raw data itself.  That confuses the graph, which is also why there's a few other things about it that don't look correct.

From memory, we captured it from an MX-800 via the Intellivue implementation, but the MX-800 produces waveforms on some bizzare data interval like 625ms, so we collect successive captures until there's a nice round number of data points over a round number of seconds.  There's an explanation of it in the code somewhere.  That was more complex than we needed for the simulated device so the frequency of the "playback" from the capture is slightly out.

Fortunately for the original project, we just needed something with waveform values that had max and min of real values, and the actual values in these waveforms are 120/80 and that was all that mattered at the time.

closed 2024-09-27 by rhepstein1