mdpnp/mdpnp #123: sample python code for waveforms

rhepstein1, 2024-09-29:
Do you have any sample python code for capturing the waveforms from an OpenIce device, similar to what you have provided in numeric_subscriber.py

rhepstein1, 2024-09-29:
> CAUTION: This email originated from outside the organization. DO NOT CLICK ON LINKS or OPEN ATTACHMENTS unless you know and trust the sender.
>
> The "samples" directory next to numeric_subscriber should do what you need.

Which file should I modify? The ice_subscriber or the ice_publisher? I assume the former to be able to retrieve data from an Ice Device connector, but please confirm.

Also, there is a comment:
# DomainParticipant QoS is configured in USER_QOS_PROFILES.xml

Do I need to do anything in that xml file to get it to work?

rhepstein1, 2024-09-30:
ice_subscriber.py worked without any modifications on my windows machine to pull in the waveform data.

rhepstein1, 2024-09-30:
I modified the code to extract selected waveform parameters, times, and values and write them to a file. The process provided is very useful for projects involving waveform signal processing.

closed 2024-09-30 by rhepstein1

simonkelly1973, 2024-09-30:
That's good to know, thanks.  We expected the Python functionality to be popular, so it's good to see it being used by other interested parties.