mdpnp/mdpnp #127: maven.restlet.org causes build failure

simonkelly1973, 2024-10-24:
We have a reference in the build scripts to maven.restlet.org .  That has previously worked, but doesn't seem to have been the up to date location for the components that we want for a long time.  Now, that reference causes numerous build errors like this

[Fatal Error] aws-java-sdk-bom-1.11.1000.pom:2:3: The markup in the document preceding the root element must be well-formed.
[Fatal Error] nashorn-core-15.4.pom:2:3: The markup in the document preceding the root element must be well-formed.
[Fatal Error] slf4j-api-2.0.3.pom:2:3: The markup in the document preceding the root element must be well-formed.
[Fatal Error] commons-cli-1.2.pom:2:3: The markup in the document preceding the root element must be well-formed.
[Fatal Error] log4j-slf4j2-impl-2.19.0.pom:2:3: The markup in the document preceding the root element must be well-formed.
[Fatal Error] log4j-api-2.19.0.pom:2:3: The markup in the document preceding the root element must be well-formed.

rhepstein1, 2024-10-24:
I am seeing these errors. does it need to be fixed or can the errors be ignored?

rhepstein1, 2024-10-24:
I updated the main build.gradle file which resolves the [Fatal Error] messages but I am still unable to compile under windows;

one of the errors is:
Project :interop-lab:demo-apps => no module-info.java found

I don't recall seeing that before.

What am I missing? Do I need to update gradlew?

Richard

C:\MDPNP\mdpnp-2.0.0>gradlew --warning-mode all :interop-lab:demo-apps:run
NOTE: To build with OpenSplice compatibility please set the OSPL_HOME and SPLICE_TARGET environment variables.

> Configure project :data-types:x73-idl-rti-dds
rtiddsgenExplodeResources started
rtiddsgenExplodeResources ended

> Configure project :interop-lab:demo-apps
Project :interop-lab:demo-apps => no module-info.java found

> Task :data-types:x73-idl-rti-dds:iceddsgenJava
The JavaExecHandleBuilder.setMain(String) method has been deprecated. This is scheduled to be removed in Gradle 8.0. Please use the mainClass property instead. Consult the upgrading guide for further information: https://docs.gradle.org/7.6/userguide/upgrading_version_7.html#java_exec_properties
        at build_3yj11kbfzy7w3upyy4e3a8g99$_run_closure8$_closure17$_closure18$_closure19.doCall(C:\MDPNP\mdpnp-2.0.0\data-types\x73-idl-rti-dds\build.gradle:147)
        (Run with --stacktrace to get the full stack trace of this deprecation warning.)
INFO com.rti.ndds.nddsgen.Main Running rtiddsgen version 4.3.0, please wait ...
ERROR com.rti.ndds.nddsgen.Main Fail:  java.lang.Exception: The preprocessor 'CL.EXE' cannot be found in your path.
If your IDL file contains preprocessor directives, be sure that 'CL.EXE'
is in your path or specify an alternative preprocessor through the use of the
flag -ppPath.
If your IDL file doesn't contain preprocessor directives you can disable the
use of the preprocessor using the flag -ppDisable
INFO com.rti.ndds.nddsgen.Main Done (failures)

> Task :data-types:x73-idl-rti-dds:iceddsgenJava FAILED

FAILURE: Build failed with an exception.

* Where:
Build file 'C:\MDPNP\mdpnp-2.0.0\data-types\x73-idl-rti-dds\build.gradle' line: 145

* What went wrong:
Execution failed for task ':data-types:x73-idl-rti-dds:iceddsgenJava'.
> Process 'command 'C:\Program Files\Java\jdk-17\bin\java.exe'' finished with non-zero exit value -1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
2 actionable tasks: 2 executed

***********************************8
************************************
Here is the gradlew info

C:\MDPNP\mdpnp-2.0.0>gradlew --version

------------------------------------------------------------
Gradle 7.6
------------------------------------------------------------

Build time:   2022-11-25 13:35:10 UTC
Revision:     daece9dbc5b79370cc8e4fd6fe4b2cd400e150a8

Kotlin:       1.7.10
Groovy:       3.0.13
Ant:          Apache Ant(TM) version 1.10.11 compiled on July 10 2021
JVM:          17.0.13 (Oracle Corporation 17.0.13+10-LTS-268)
OS:           Windows 10 10.0 amd64

rhepstein1, 2024-10-24:
I am getting other errors on my original windows machine after correcting the maven reference.
These also appear related to maven

> Configure project :interop-lab:demo-apps
Project :interop-lab:demo-apps => no module-info.java found

There are also scores of preprocessor directive not supported even though I have CL.EXE in my path
And hundreds of warnings related to JavaFX

At the end of all this, the build is reported as successful  and the ICE Supervisor opens and works, but WHY are all these warnings and errors being thrown.

This was not the case previously.
The program needs to compile without errors to provide reassurance to users that it is working as designed.

simonkelly1973, 2024-10-25:
> I am getting other errors on my original windows machine after correcting the maven reference. These also appear related to maven
>
>
> There are also scores of preprocessor directive not supported even [...]

Could you provide a full build log so we can investigate?  You can produce that by doing something like

.\gradlew clean compileJava > build.log 2>&1

After that just attach the build.log to this ticket and we'll take a look.

Thanks

rhepstein1, 2024-10-25:
I downloaded the latest files from the mdpnp github repository.

When I compile on Windows with gradlew :interop-lab:demo-apps:run
I get the message in stdout:
Project :interop-lab:demo-apps => no module-info.java found
There are warnings about the preprocessor directive supported even though I have the folder with cl.exe in my path.
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.41.34120\bin\Hostx64\x86
There are lots of warnings about deprecated features but those don't appear to matter since the project compiles successfully and the ICE Supervisor works.

So should I just ignore the message  about the missing module-info.java?

It seems like everything may be OK.

[build.log](https://github.com/user-attachments/files/17522305/build.log)

simonkelly1973, 2024-10-25:
Yes, the "modular system" that the "missing module" message refers to is a feature of the Java programming language that OpenICE doesn't use yet.  We have it in mind to adopt, but OpenICE has many third party dependencies that aren't compatible with that (or at least, were not compatible last time we checked).

There are a couple of other trivial deprecation warnings, but apart from that this does look like a normal build log.

rhepstein1, 2024-10-25:
Thanks. I updated the files on my raspberry pi to the latest build and that is working fine. There are a lot of moving pieces so I think it is wise to be very conservative about implementing "features."

simonkelly1973, 2024-10-25:
Yes, it's an interesting balance.  Doing nothing (in terms of using newer versions of languages and frameworks) does make things easy in terms of allowing you to concentrate on new OpenICE devices and applications, but it does make it quite hard when you are finally "forced" to upgrade because your current platform "doesn't work on the latest version of Windows" or something.  Then you find you're many releases behind and it's a big leap to encompass all the changes that are required...