mdpnp/mdpnp #129: javafx version not updated in the current build

rhepstein1, 2024-10-29:
It appears that the current default branch may have the wrong javafx versions in:

C:\MDPNP\mdpnp-2.0.0\interop-lab\demo-guis-javafx\build.gradle
C:\MDPNP\mdpnp-2.0.0\interop-lab\demo-apps\build.gradle

I think those should be 17.0.8 not 17.0.11

If the 17.0.11 version now works (it didn't previously), please advise.

Thanks.

simonkelly1973, 2024-10-29:
Well - that all depends on which platform you are referring to.  The issue with the JavaFX components is that they are platform/architecture specific.  We rely on those being available, rather than us building those components ourselves.  The issue with 17.0.11 is that it was available on the "main" platforms that we used, for Windows on x86/64 bit, OS/X for Mac and so on.  However, that version was not built for the Raspberry Pi.  When we discovered that, the "fix" was to rever back to 17.0.8 that was the latest version that was built for Raspberry Pi.  But that workaround isn't necessary on any other platform - and that's really why that small update hasn't made it into the main repository.

rhepstein1, 2024-10-29:
Is it possible to either conditionally implement the JavaFX version based on determination of the OS or provide both versions in the download with a bash script that will delete the non-RPi version and rename the RPi version so that developers/user don't have to manually edit the files. The latter requirement is a burdensome task isn't something that is typically required for source files.

closed 2024-11-12 by rhepstein1

rhepstein1, 2025-08-15:
> CAUTION: This email originated from outside the organization. DO NOT CLICK ON LINKS or OPEN ATTACHMENTS unless you know and trust the sender.

Simon:

I’m back after a pause working on getting OpenIce ported to a RPi platform with a small touchscreen to create a standalone, portable means of capturing the physiologic data from patient monitors and processing it in python.

I dropped an “issue” on GitHub, but am not sure if this is being monitored, as there haven’t been any questions raised by users since my last post on Nov 25, 2024.

Basically, I’m looking for the MDC code that contains the ETCO2 values shown in the simulation software. I tried all the ones I could find in the Philips MDC documentation, but couldn’t get any to work. Perhaps it’s not available in the simulation software you wrote?

Not mentioned, I also need the MAP, but I don’t think that is being exported, just the systolic and diastolic BP. For test purposes, I’m just calculating the MAP from the SBP and DBP.
But if there is an MDC that is being sent, that would be better. I’ve not connected yet to a Philips Monitor, but that should work fine as I have VitalRecorded working to capture all the data so the cable I built and the configuration parameters I set on the Philips are correct.

Rich